kenzok8/openwrt-packages #498: 编译adgardhome时出错

Kam323, 2024-09-22:
make[3]: *** [Makefile:84: /home/test/lede/build_dir/target-x86_64_musl/AdGuardHome-0.107.52/.built] Error 1
make[3]: Leaving directory '/home/test/lede/feeds/kenzo/adguardhome'
time: package/feeds/kenzo/adguardhome/compile#0.23#1.15#1.29
    ERROR: package/feeds/kenzo/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/adguardhome/compile] Error 1
make[2]: Leaving directory '/home/test/lede'
make[1]: *** [package/Makefile:110: /home/test/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/test/lede'
make: *** [/home/test/lede/include/toplevel.mk:231：world] 错误 2

Kam323, 2024-09-22:
更换openwrt的golang版本，正在尝试重新编译，用的是以下命令：
rm -rf feeds/packages/lang/golang
git clone https://github.com/kenzok8/golang feeds/packages/lang/golang

Kam323, 2024-09-23:
ubuntu22.04.5也安装了go1.23rc2，编译adguard依旧出错，这个意思是访问golang.org timeout吗？我已经是科学上网环境了

../../internal/aghalg/nullbool.go:8:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghalg/aghalg.go:10:2: golang.org/x/exp@v0.0.0-20240604190554-fc45aab8b7f8: Get "https://proxy.golang.org/golang.org/x/exp/@v/v0.0.0-20240604190554-fc45aab8b7f8.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/version/version.go:12:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghhttp/aghhttp.go:10:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghhttp/aghhttp.go:11:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghos/filewalker.go:8:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghos/filewalker.go:9:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghos/fswatcher.go:12:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghos/fswatcher.go:13:2: github.com/fsnotify/fsnotify@v1.7.0: Get "https://proxy.golang.org/github.com/fsnotify/fsnotify/@v/v1.7.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghos/os_unix.go:9:2: golang.org/x/sys@v0.21.0: Get "https://proxy.golang.org/golang.org/x/sys/@v/v0.21.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/net.go:17:2: github.com/AdguardTeam/dnsproxy@v0.71.2: Get "https://proxy.golang.org/github.com/%21adguard%21team/dnsproxy/@v/v0.71.2.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/hostscontainer.go:13:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/dhcp_unix.go:15:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/ignore.go:7:2: github.com/AdguardTeam/urlfilter@v0.19.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/urlfilter/@v/v0.19.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/ignore.go:8:2: github.com/AdguardTeam/urlfilter@v0.19.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/urlfilter/@v/v0.19.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/net_linux.go:17:2: github.com/google/renameio/v2@v2.0.0: Get "https://proxy.golang.org/github.com/google/renameio/v2/@v/v2.0.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/dhcp_unix.go:16:2: github.com/insomniacslk/dhcp@v0.0.0-20240419123447-f1cffa2c0c49: Get "https://proxy.golang.org/github.com/insomniacslk/dhcp/@v/v0.0.0-20240419123447-f1cffa2c0c49.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/interfaces_linux.go:9:2: github.com/insomniacslk/dhcp@v0.0.0-20240419123447-f1cffa2c0c49: Get "https://proxy.golang.org/github.com/insomniacslk/dhcp/@v/v0.0.0-20240419123447-f1cffa2c0c49.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/dhcp_unix.go:17:2: github.com/insomniacslk/dhcp@v0.0.0-20240419123447-f1cffa2c0c49: Get "https://proxy.golang.org/github.com/insomniacslk/dhcp/@v/v0.0.0-20240419123447-f1cffa2c0c49.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/dhcp_unix.go:18:2: github.com/insomniacslk/dhcp@v0.0.0-20240419123447-f1cffa2c0c49: Get "https://proxy.golang.org/github.com/insomniacslk/dhcp/@v/v0.0.0-20240419123447-f1cffa2c0c49.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghnet/dhcp_unix.go:19:2: github.com/insomniacslk/dhcp@v0.0.0-20240419123447-f1cffa2c0c49: Get "https://proxy.golang.org/github.com/insomniacslk/dhcp/@v/v0.0.0-20240419123447-f1cffa2c0c49.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/aghrenameio/renameio_unix.go:8:2: github.com/google/renameio/v2@v2.0.0: Get "https://proxy.golang.org/github.com/google/renameio/v2/@v/v2.0.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/filtering/rulelist/filter.go:16:2: github.com/AdguardTeam/golibs@v0.24.0: Get "https://proxy.golang.org/github.com/%21adguard%21team/golibs/@v/v0.24.0.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/filtering/rulelist/engine.go:13:2: github.com/c2h5oh/datasize@v0.0.0-20231215233829-aa82cc1e6500: Get "https://proxy.golang.org/github.com/c2h5oh/datasize/@v/v0.0.0-20231215233829-aa82cc1e6500.zip": dial tcp 142.251.215.241:443: i/o timeout
../../internal/filtering/rulelist/rulelist.go:11:2: github.com/google/uuid@v1.6.0: Get "https://proxy.golang.org/github.com/google/uuid/@v/v1.6.0.zip": dial tcp 142.251.215.241:443: i/o timeout
[... cut ...]
make[3]: Leaving directory '/home/test/lede/feeds/kenzo/adguardhome'
time: package/feeds/kenzo/adguardhome/compile#0.32#1.57#781.99
    ERROR: package/feeds/kenzo/adguardhome failed to build.
make[2]: *** [package/Makefile:116: package/feeds/kenzo/adguardhome/compile] Error 1
make[2]: Leaving directory '/home/test/lede'
make[1]: *** [package/Makefile:110: /home/test/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/test/lede'
make: *** [/home/test/lede/include/toplevel.mk:231：world] 错误 2

kenzok8, 2024-09-27:
你网络的问题吧。链接你随便点下，看看能下载吗？如果是vps或者自己的虚拟机编译的，ssh下你用`wget https://%^#%^#`看下能否download

dghuig, 2024-10-04:
一样的报错！一开始我也以为是golang版本问题！结果是新版v0.107.53的问题，回退到v0.107.52就正常编译！

closed 2026-02-24 by kenzok8