kenzok8/openwrt-packages #505: 编译报错23.05

lexin8, 2024-11-09:
编译报错luci-23.05版本

SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/openssh-without-pam/openssh-9.8p1/.pkgdir/openssh-keygen/. /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.openssh-keygen_installed
echo "openssh-keygen" >> /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/openssh.without-pam.install
mkdir -p /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/openssh-without-pam/openssh-9.8p1/.pkgdir/openssh-sftp-server/. /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.openssh-sftp-server_installed
echo "openssh-sftp-server" >> /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/openssh.without-pam.install
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/openssh'
time: package/feeds/packages/openssh/without-pam/compile#0.18#0.06#0.22
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/parted'
mkdir -p /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/.pkgdir/libparted/. /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.libparted_installed
echo "libparted" >> /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/parted.default.install
mkdir -p /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/parted-3.6/.pkgdir/parted/. /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/'
touch /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/root-mediatek/stamp/.parted_installed
echo "parted" >> /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/pkginfo/parted.default.install
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/parted'
time: package/feeds/packages/parted/compile#0.16#0.07#0.21
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/libxcrypt'
rm -f /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36/.built
touch /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36/.built_check
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36=libxcrypt-4.4.36 -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36=libxcrypt-4.4.36 -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -znow -zrelro " make  -C /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36=libxcrypt-4.4.36 -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36'
make  all-am
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36'
aarch64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I.  -I./lib -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include  -Wall -Wextra -Walloc-zero -Walloca -Wbad-function-cast -Wcast-align -Wcast-qual -Wconversion -Wformat=2 -Wformat-overflow=2 -Wformat-signedness -Wformat-truncation=1 -Wlogical-op -Wmissing-declarations -Wmissing-prototypes -Wnested-externs -Wnull-dereference -Wold-style-definition -Wpointer-arith -Wrestrict -Wshadow -Wstrict-overflow=2 -Wstrict-prototypes -Wundef -Wvla -Wwrite-strings -Wpedantic -Werror -fno-plt -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36=libxcrypt-4.4.36 -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro  -c -o lib/gen-des-tables.o lib/gen-des-tables.c
In file included from lib/gen-des-tables.c:52:
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify/stdio.h: In function 'snprintf':
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify/stdio.h:101:9: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  101 |         return __orig_snprintf(__s, __n, __f, __builtin_va_arg_pack());
      |         ^~~~~~
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify/stdio.h: In function 'sprintf':
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify/stdio.h:110:17: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  110 |                 __r = __orig_snprintf(__s, __b, __f, __builtin_va_arg_pack());
      |                 ^~~
/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify/stdio.h:114:17: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  114 |                 __r = __orig_sprintf(__s, __f, __builtin_va_arg_pack());
      |                 ^~~
cc1: all warnings being treated as errors
make[5]: *** [Makefile:1915: lib/gen-des-tables.o] Error 1
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36'
make[4]: *** [Makefile:1349: all] Error 2
make[3]: *** [Makefile:58: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/libxcrypt-4.4.36/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libxcrypt'
time: package/feeds/packages/libxcrypt/compile#0.22#0.06#0.26
    ERROR: package/feeds/packages/libxcrypt failed to build.
make[2]: *** [package/Makefile:129: package/feeds/packages/libxcrypt/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:123: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2

kenzok8, 2024-11-10:
错误是在编译 libxcrypt 包时出现的,具体要看哪个依赖包的

gyssi007, 2024-11-14:
我也遇到同样问题了

gyssi007, 2024-11-15:
修改下面文件
package/feeds/packages/libxcrypt/Makefile
插入一行内容如下：

PKG_FORTIFY_SOURCE=0

这样就解决了

kenzok8, 2026-02-24:
自动巡检：该 Issue 长时间未更新（>=180 天），先关闭以便集中处理当前问题。

如问题仍存在，请按最新源码重新复现后提交新 Issue，并附：
- 固件版本与目标架构
- 完整编译/运行日志（从第一个报错开始）
- 复现步骤

Auto triage: this issue has been inactive for >=180 days, so it is being closed for backlog cleanup.
If the issue still exists, please open a new issue with updated logs and clear reproduction steps.