kenzok8/openwrt-packages #517: 这个两天编译错误，麻烦大佬帮忙看看。

LIUBANGJIAN, 2025-01-19:
2025-01-19T10:35:08.5429732Z   345 |             free (root);
2025-01-19T10:35:08.5429994Z       |                   ^~~~
2025-01-19T10:35:08.5430317Z In file included from ./include/grub/emu/misc.h:25,
2025-01-19T10:35:08.5430665Z                  from util/setup.c:22:
2025-01-19T10:35:08.5431201Z ./grub-core/lib/gnulib/stdlib.h:739:1: note: expected 'void *' but argument is of type 'const char *'
2025-01-19T10:35:08.5431797Z   739 | _GL_FUNCDECL_RPL (free, void, (void *ptr));
2025-01-19T10:35:08.5432127Z       | ^~~~~~~~~~~~~~~~
2025-01-19T10:35:08.6405378Z mv -f util/.deps-util/grub_install-setup_sparc.Tpo util/.deps-util/grub_install-setup_sparc.Po
2025-01-19T10:35:08.6461594Z x86_64-openwrt-linux-musl-gcc -std=gnu99 -fno-common  -fno-builtin-gettext -Wall -W -Wshadow -Wpointer-arith -Wundef -Wchar-subscripts -Wcomment -Wdeprecated-declarations -Wdisabled-optimization -Wdiv-by-zero -Wfloat-equal -Wformat-extra-args -Wformat-security -Wformat-y2k -Wimplicit -Wimplicit-function-declaration -Wimplicit-int -Wmain -Wmissing-braces -Wmissing-format-attribute -Wmultichar -Wparentheses -Wreturn-type -Wsequence-point -Wshadow -Wsign-compare -Wswitch -Wtrigraphs -Wunknown-pragmas -Wunused -Wunused-function -Wunused-label -Wunused-parameter -Wunused-value  -Wunused-variable -Wwrite-strings -Wnested-externs -Wstrict-prototypes -Wcast-align  -Wextra -Wattributes -Wendif-labels -Winit-self -Wint-to-pointer-cast -Winvalid-pch -Wmissing-field-initializers -Wnonnull -Woverflow -Wvla -Wpointer-to-int-cast -Wstrict-aliasing -Wvariadic-macros -Wvolatile-register-var -Wpointer-sign -Wmissing-include-dirs -Wmissing-prototypes -Wmissing-declarations -Wformat=2 -Wno-undef -Wno-sign-compare -Wno-unused -Wno-unused-parameter -Wno-redundant-decls -Wno-unreachable-code -Wno-conversion -Wno-error=attributes  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.12=grub-2.12 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro     -L/home/runner/work/lede/lede/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/runner/work/lede/lede/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro  -o grub-install grub-core/disk/grub_install-host.o grub-core/kern/emu/grub_install-argp_common.o grub-core/kern/emu/grub_install-hostfs.o grub-core/lib/grub_install-reed_solomon.o grub-core/osdep/grub_install-blocklist.o grub-core/osdep/grub_install-compress.o grub-core/osdep/grub_install-config.o grub-core/osdep/grub_install-init.o grub-core/osdep/grub_install-ofpath.o grub-core/osdep/grub_install-platform.o grub-core/osdep/grub_install-platform_unix.o grub-core/osdep/grub_install-random.o util/grub_install-config.o util/grub_install-editenv.o util/grub_install-grub-install-common.o util/grub_install-grub-install.o util/grub_install-grub-mkimage32.o util/grub_install-grub-mkimage64.o util/grub_install-mkimage.o util/grub_install-probe.o util/grub_install-render-label.o util/grub_install-resolve.o util/grub_install-setup_bios.o util/grub_install-setup_sparc.o   libgrubmods.a libgrubgcry.a libgrubkern.a grub-core/lib/gnulib/libgnu.a        
2025-01-19T10:35:08.6792537Z (for x in util/grub-kbdcomp.in ; do cat ./"$x"; done) | ./config.status --file=grub-kbdcomp:-
2025-01-19T10:35:08.7192650Z config.status: creating grub-kbdcomp
2025-01-19T10:35:08.7319022Z chmod a+x grub-kbdcomp
2025-01-19T10:35:08.7339086Z (for x in util/grub.d/00_header.in ; do cat ./"$x"; done) | ./config.status --file=00_header:-
2025-01-19T10:35:08.7740509Z config.status: creating 00_header
2025-01-19T10:35:08.7868986Z chmod a+x 00_header
2025-01-19T10:35:08.7888500Z (for x in util/grub.d/10_linux.in ; do cat ./"$x"; done) | ./config.status --file=10_linux:-
2025-01-19T10:35:08.8287119Z config.status: creating 10_linux
2025-01-19T10:35:08.8413584Z chmod a+x 10_linux
2025-01-19T10:35:08.8433206Z (for x in util/grub.d/20_linux_xen.in ; do cat ./"$x"; done) | ./config.status --file=20_linux_xen:-
2025-01-19T10:35:08.8831868Z config.status: creating 20_linux_xen
2025-01-19T10:35:08.8958864Z chmod a+x 20_linux_xen
2025-01-19T10:35:08.8977986Z (for x in util/grub.d/25_bli.in ; do cat ./"$x"; done) | ./config.status --file=25_bli:-
2025-01-19T10:35:08.9376038Z config.status: creating 25_bli
2025-01-19T10:35:08.9512955Z chmod a+x 25_bli
2025-01-19T10:35:08.9532376Z (for x in util/grub.d/30_os-prober.in ; do cat ./"$x"; done) | ./config.status --file=30_os-prober:-
2025-01-19T10:35:08.9932606Z config.status: creating 30_os-prober
2025-01-19T10:35:09.0060793Z chmod a+x 30_os-prober
2025-01-19T10:35:09.0080183Z (for x in util/grub.d/30_uefi-firmware.in ; do cat ./"$x"; done) | ./config.status --file=30_uefi-firmware:-
2025-01-19T10:35:09.0480212Z config.status: creating 30_uefi-firmware
2025-01-19T10:35:09.0603909Z chmod a+x 30_uefi-firmware
2025-01-19T10:35:09.0623302Z (for x in util/grub.d/40_custom.in ; do cat ./"$x"; done) | ./config.status --file=40_custom:-
2025-01-19T10:35:09.1023325Z config.status: creating 40_custom
2025-01-19T10:35:09.1158885Z chmod a+x 40_custom
2025-01-19T10:35:09.1178478Z (for x in util/grub.d/41_custom.in ; do cat ./"$x"; done) | ./config.status --file=41_custom:-
2025-01-19T10:35:09.1576347Z config.status: creating 41_custom
2025-01-19T10:35:09.1711704Z chmod a+x 41_custom
2025-01-19T10:35:09.1731044Z (for x in util/grub-mkconfig_lib.in ; do cat ./"$x"; done) | ./config.status --file=grub-mkconfig_lib:-
2025-01-19T10:35:09.2127837Z config.status: creating grub-mkconfig_lib
2025-01-19T10:35:09.2253276Z chmod a+x grub-mkconfig_lib
2025-01-19T10:35:09.2272238Z (for x in tests/util/grub-shell.in ; do cat ./"$x"; done) | ./config.status --file=grub-shell:-
2025-01-19T10:35:09.2670496Z config.status: creating grub-shell
2025-01-19T10:35:09.2802060Z chmod a+x grub-shell
2025-01-19T10:35:09.2821763Z (for x in tests/util/grub-shell-tester.in ; do cat ./"$x"; done) | ./config.status --file=grub-shell-tester:-
2025-01-19T10:35:09.3219187Z config.status: creating grub-shell-tester
2025-01-19T10:35:09.3343279Z chmod a+x grub-shell-tester
2025-01-19T10:35:09.3362261Z (for x in tests/util/grub-shell-luks-tester.in ; do cat ./"$x"; done) | ./config.status --file=grub-shell-luks-tester:-
2025-01-19T10:35:09.3761341Z config.status: creating grub-shell-luks-tester
2025-01-19T10:35:09.3885923Z chmod a+x grub-shell-luks-tester
2025-01-19T10:35:09.3905962Z gcc -o garbage-gen -std=gnu99 -fno-common  -DGRUB_FILE=\"util/garbage-gen.c\" -I. -I. -I. -I. -I./include -I./include -I./grub-core/lib/libgcrypt-grub/src/   util/garbage-gen.c
2025-01-19T10:35:09.4271255Z (for x in tests/util/grub-fs-tester.in ; do cat ./"$x"; done) | ./config.status --file=grub-fs-tester:-
2025-01-19T10:35:09.4669091Z config.status: creating grub-fs-tester
2025-01-19T10:35:09.4821714Z chmod a+x grub-fs-tester
2025-01-19T10:35:09.4840977Z (for x in util/grub-mkconfig.in ; do cat ./"$x"; done) | ./config.status --file=grub-mkconfig:-
2025-01-19T10:35:09.5240889Z config.status: creating grub-mkconfig
2025-01-19T10:35:09.5367078Z chmod a+x grub-mkconfig
2025-01-19T10:35:09.5386294Z (for x in util/grub-set-default.in ; do cat ./"$x"; done) | ./config.status --file=grub-set-default:-
2025-01-19T10:35:09.5791771Z config.status: creating grub-set-default
2025-01-19T10:35:09.5916716Z chmod a+x grub-set-default
2025-01-19T10:35:09.5935746Z (for x in util/grub-reboot.in ; do cat ./"$x"; done) | ./config.status --file=grub-reboot:-
2025-01-19T10:35:09.6336588Z config.status: creating grub-reboot
2025-01-19T10:35:09.6463530Z chmod a+x grub-reboot
2025-01-19T10:35:09.6534333Z make[6]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.12'
2025-01-19T10:35:09.6545397Z Making all in po
2025-01-19T10:35:09.6572977Z make[6]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.12/po'
2025-01-19T10:35:09.6573920Z make[6]: Nothing to be done for 'all'.
2025-01-19T10:35:09.6574551Z make[6]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.12/po'
2025-01-19T10:35:09.6576704Z Making all in docs
2025-01-19T10:35:09.6613745Z make[6]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.12/docs'
2025-01-19T10:35:09.6614580Z make[6]: Nothing to be done for 'all'.
[... cut ...]
2025-01-19T10:35:10.2245583Z [command]/usr/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :"
2025-01-19T10:35:10.2524618Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2025-01-19T10:35:10.2560390Z [command]/usr/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :"
2025-01-19T10:35:10.2956365Z Cleaning up orphan processes

这个两天编译错误，麻烦大佬帮忙看看。

smmya, 2025-01-21:
+1，这段时间编译一直出问题

heiljsf, 2025-03-20:
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'pciutils', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'wsdd2', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-ksmbd', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba4', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_dnspod/Makefile' has a dependency on 'wget-ssl', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/qat/libs/eudev/Makefile' has a dependency on 'libkmod', which does not exist
WARNING: Makefile 'package/qat/libs/eudev/Makefile' has a build dependency on 'gperf/host', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'bash', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'jq', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'ntpdate', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'smartmontools', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'zoneinfo-all', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'coreutils-install', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'coreutils-df', which does not exist
WARNING: Makefile 'package/lean/fogvdn/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/libs/gettext-full/Makefile' has a build dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/libs/gettext-full/Makefile' has a build dependency on 'libxml2', which does not exist
WARNING: Makefile 'package/libs/gettext-full/Makefile' has a build dependency on 'gperf/host', which does not exist
WARNING: Makefile 'package/libs/gettext-full/Makefile' has a build dependency on 'libunistring/host', which does not exist
WARNING: Makefile 'package/libs/gettext-full/Makefile' has a build dependency on 'libxml2/host', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/libs/libselinux/Makefile' has a dependency on 'libpcre2', which does not exist
WARNING: Makefile 'package/libs/libselinux/Makefile' has a build dependency on 'pcre2/host', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-fogvdn/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/lean/luci-app-fogvdn/Makefile' has a dependency on 'jq', which does not exist
WARNING: Makefile 'package/lean/luci-app-fogvdn/Makefile' has a dependency on 'luci-lua-runtime', which does not exist
WARNING: Makefile 'package/lean/luci-app-fogvdn/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-fogvdn/Makefile' has a build dependency on 'csstidy/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-fogvdn/Makefile' has a build dependency on 'luasrcdiet/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-leigod-acc/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/lean/luci-app-leigod-acc/Makefile' has a dependency on 'miniupnpd', which does not exist
WARNING: Makefile 'package/lean/luci-app-leigod-acc/Makefile' has a dependency on 'conntrack', which does not exist
WARNING: Makefile 'package/lean/luci-app-leigod-acc/Makefile' has a dependency on 'conntrackd', which does not exist
WARNING: Makefile 'package/lean/luci-app-leigod-acc/Makefile' has a dependency on 'jq', which does not exist
WARNING: Makefile 'package/lean/luci-app-leigod-acc/Makefile' has a dependency on 'luci-lua-runtime', which does not exist
[... cut ...]
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/firmware/quantenna/Makefile' has a dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/firmware/quantenna/Makefile' has a build dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/network/services/samba36/Makefile' has a dependency on 'libaio', which does not exist