kenzok8/openwrt-packages #536: 最新的golang编译失败

norman-arch, 2025-09-18:
`make[3]: Entering directory '/home/norman/immortalwrt-bpi4/feeds/packages/lang/golang/golang'
tar -C "/home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1" --strip-components=1 -xzf "/home/norman/immortalwrt-bpi4/dl/go1.25.1.src.tar.gz"
[ ! -d ./src/ ] || cp -fpR ./src/* /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1
mkdir -p "/home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.go_bootstrap"
tar -C "/home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.go_bootstrap" --strip-components=1 -xzf "/home/norman/immortalwrt-bpi4/dl/go1.20.6.linux-amd64.tar.gz"
touch /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.prepared880ee58da12773fb0e9ad13e5d0e91c3_6664517399ebbbc92a37c5bb081b5c53
(cd /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/; if [ -x configure ]; then cp -fpR /home/norman/immortalwrt-bpi4/scripts/config.{guess,sub} /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1// && CC="/home/norman/immortalwrt-bpi4/staging_dir/host/bin/gcc" CFLAGS="-O2 -I/home/norman/immortalwrt-bpi4/staging_dir/host/include -I/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/include -I/home/norman/immortalwrt-bpi4/staging_dir/target-aarch64_cortex-a53_musl/host/include" CXX="/home/norman/immortalwrt-bpi4/staging_dir/host/bin/g++" CPPFLAGS="-I/home/norman/immortalwrt-bpi4/staging_dir/host/include -I/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/include -I/home/norman/immortalwrt-bpi4/staging_dir/target-aarch64_cortex-a53_musl/host/include" CXXFLAGS="-O2 -I/home/norman/immortalwrt-bpi4/staging_dir/host/include -I/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/include -I/home/norman/immortalwrt-bpi4/staging_dir/target-aarch64_cortex-a53_musl/host/include" LDFLAGS="-L/home/norman/immortalwrt-bpi4/staging_dir/host/lib -L/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/lib -L/home/norman/immortalwrt-bpi4/staging_dir/target-aarch64_cortex-a53_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/home/norman/immortalwrt-bpi4/staging_dir/hostpkg --exec-prefix=/home/norman/immortalwrt-bpi4/staging_dir/hostpkg --sysconfdir=/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/etc --localstatedir=/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/var --sbindir=/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/bin ; fi )
touch /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.configured
mkdir -p "/home/norman/immortalwrt-bpi4/tmp/go-build"
( cd "/home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/src" ; GOROOT_FINAL="/home/norman/immortalwrt-bpi4/staging_dir/hostpkg/lib/go-cross"  GOROOT_BOOTSTRAP="/home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.go_bootstrap" GO_LDFLAGS="-buildmode pie" GOCACHE="/home/norman/immortalwrt-bpi4/tmp/go-build" GOENV=off CC="/home/norman/immortalwrt-bpi4/staging_dir/host/bin/gcc" CXX="/home/norman/immortalwrt-bpi4/staging_dir/host/bin/g++"  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.go_bootstrap. (go1.20.6 linux/amd64)
found packages main (build.go) and building_Go_requires_Go_1_22_6_or_later (notgo122.go) in /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/src/cmd/dist
cmd/dist/buildtool.go:16:2: cannot find package "go/version" in any of:
        /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.go_bootstrap/src/go/version (from $GOROOT)
        /home/norman/go/src/go/version (from $GOPATH)
cmd/dist/build.go:19:2: cannot find package "slices" in any of:
        /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.go_bootstrap/src/slices (from $GOROOT)
        /home/norman/go/src/slices (from $GOPATH)
make[3]: *** [Makefile:379: /home/norman/immortalwrt-bpi4/build_dir/hostpkg/go-1.25.1/.built] Error 1
make[3]: Leaving directory '/home/norman/immortalwrt-bpi4/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#1.67#0.62#2.94
    ERROR: package/feeds/packages/golang [host] failed to build.
make[2]: *** [package/Makefile:181: package/feeds/packages/golang/host/compile] Error 1
make[2]: Leaving directory '/home/norman/immortalwrt-bpi4'
make[1]: *** [package/Makefile:177: /home/norman/immortalwrt-bpi4/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/norman/immortalwrt-bpi4'
make: *** [/home/norman/immortalwrt-bpi4/include/toplevel.mk:233: world] Error 2
`

kenzok8, 2025-09-18:
你的问题不是 Go 1.25.1 本身有 bug，而是 bootstrap Go 太旧（1.20.6）
去下面目录查看golang是否替换更新了`feeds/packages/lang/golang/golang/Makefile`
另外试着删除旧的golang `rm -rf dl/go*.tar.gz
`

norman-arch, 2025-09-20:
解决了，谢谢